arm/arm-toolchain #87: [Downstream Change] `sincos()` vectorization

MacDue, 2025-02-11:
This issue tracks the downstream changes needed to enable `sincos()` vectorization in the Arm toolchain. These changes are only for the release 20.x branch (https://github.com/arm/arm-toolchain/tree/release/arm-software/20.x) as we intend to complete up-streaming the changes for LLVM 21. 

- [ ] https://github.com/arm/arm-toolchain/pull/82
- [ ] https://github.com/arm/arm-toolchain/pull/83
- [ ] https://github.com/arm/arm-toolchain/pull/84 
     - Behavior change: `sincos()` becomes vectorizable with `-fveclib=ArmPL/SLEEF`

The patches were not possible to upstream for LLVM 20 due to long review times.

MacDue, 2025-02-12:
> Can you also provide links to the llvm-project main branch patches? Have these landed or are they in review?

Added the upstream PR links (above). The upstream patches have not landed.

MacDue, 2025-02-27:
Update: All patches have landed upstream (so should be available in the 21.x release without any downstream changes).

voltur01, 2026-10-21:
One off release branch downstream change - closing.

closed 2026-10-21 by voltur01