arm/arm-toolchain #177: [Downstream Change] Some test cases are failing due to the nature of time and space

pawosm-arm, 2025-03-07:
There are test cases which are very sensitive to the shared machines load. These are:

`llvm/utils/lit/tests/googletest-timeout.py`
`llvm/utils/lit/tests/max-time.py`
`llvm/utils/lit/tests/xunit-output-report-failures-only.py`
`llvm/utils/lit/tests/xunit-output.py`

As @DavidSpickett has summarized it to us:

> The issue is not speed, it's load on the machine. If you always run to the maximum capacity these tests will always be flaky, it's not a problem with the test it's a problem with the nature of time and space.
>
> The solution is not to ninja `use all the threads in the world` and not to run multiple test runs on the same instance. Leave some headroom for tests like this that spawn more threads.
>
> How much? How long is a piece of string? But some headroom anyway.
>
> Speed can help but with all the speed in the world, it would still be flakey a non-zero amount of times if you're running at and past the machine's capacity.
>
> So you have something you can tune downstream but if you really want to solve this generally, you could have some kind of "please run this at the end of the run when we probably have more resource" annotation. But this is upstream work and it does not solve the issue of shared hosts.
>
> A third option is to disable these tests downstream. The risk is very low given that they are testing the testing tool's ability to manage tests (very meta).
>
> So the thing you might have happen is some test will loop and it doesn't get killed, this would be very easy for an engineer to spot. Especially if your pipelines have a global timeout anyway (if not, great idea to add one!). Plus, if lit somehow forgot how to kill tests, upstream would have caught this because they will run this test.
>
> ...so whichever takes your fancy really. I'm not going to say one is better because I don't know the infrastructure.
>
> The other killer for shared machines is not just amount of resource, but fluctuating level of resource. This can really mess up ninja and lit.
>
> If you want to fix this right now, disabling the test downstream is the pick. Then you can come up with something more principled later, or forget all about it - I'm not judging.
>

DavidSpickett, 2025-03-10:
For `llvm/utils/lit/tests/googletest-timeout.py`, here is more about the test if you want to debate running it or not:
* It checks that lit can enforce a timeout on a googletest binary.
* It does this by starting such a binary, which then stalls, and lit should stop it before the binary finishes normally.
* If lit is stalled for a long time because it could not get a core to run on, the test binary might complete, or lit will have enforced a timeout but a greater one than expected, which is also a failure.

So if you were to not run this test in this project:
* A googletest binary could stall forever in your testing.
  * Unless you have a global job timeout, which is a good idea to have anyway.
  * A test looping should be easy for an engineer to spot, at least that *some* test is looping, sometimes it's hard to tell what.
  * You are very unlikely to modify lit downstream, so if it broke, it was likely done by upstream who *do* run this test and would therefore be fixing the issue for you.
* The lit built here may not have the ability to enforce timeouts onto googletest.
  * This is slightly annoying for any local developer that manages to produce an infinite loop. Then again, the default timeout is 5 minutes anyway, most people will ctrl-c before that limit.
  * "customers" of this toolchain likely don't consider lit to be part of the product, so you're not really guaranteeing them any features anyway.

Disabling it is downstream diff, but it would only be a one line change and easily merged.

(also downstream discussion has discovered that this requires the Python psutil module, so you could instead not install that in your own infrastructure)

If you are thinking about it in terms of costs:
* X is the value the lit feature gives you.
* Y is the cost of an engineer having to look at a false negative.
* Z is the cost of the timeout feature being broken.

I would suggest that Y is much greater than X or Z because human time is more costly than machine time. And that Z is a small number in this case because CI jobs will have a global timeout anyway and customers are not being sold lit as a product.

(if this feature was something obvious like `armclang --version`, that would be very embarrassing to break, Z would be larger)

pawosm-arm, 2025-03-31:
Closing it, as we are handling this issue in our CI.

closed 2025-03-31 by pawosm-arm