arm/arm-toolchain #297: [ATfE] Windows ZIP release artifact filename contains x86_64, but binaries are 32-bit

malsyned, 2025-04-22:
The binaries that were part of the LLVM-ET-Arm package for Windows-x86_64 were true 64-bit binaries. The ones that are part of ATfE represent that they are 64-bit, but are actually 32-bit. I noticed this when the installation instructions I had written up for LLVM-ET-Arm didn't work for ATfE. They failed to install a working toolchain because they instructed the user to install the X64 Visual C++ Redistributable, when the X32 version was what was needed.

### Actual:

```
C:\Users\VBoxUser>llvm-objdump -x C:\Users\VBoxUser\Desktop\ATfE-20.1.0-Windows-x86_64\bin\clang.exe | C:\msys64\usr\bin\grep architecture:
architecture: i386
```

### Expected:

```
architecture: x86_64
```

### Proposed fix

Either change the zip artifact filename to say "x86" or "i386" or something instead of "x86_64", or adjust the build process to actually produce 64-bit binaries.

voltur01, 2025-04-23:
Good morning, thank you for reporting this! The binaries must be 64-bit, however looks like we missed an option somewhere - we will look into this with priority to fix for LLVM 21 release.

vrukesh, 2025-05-29:
Thanks to @statham-arm , we have fix for this issue.

- Release20 branch: https://github.com/arm/arm-toolchain/pull/348
- arm-software branch: https://github.com/arm/arm-toolchain/pull/346

Kindly rebuild with the patches and check.
-Vrukesh

closed 2025-05-29 by vrukesh