arm/arm-toolchain #321: The generated ELF is broken

mikisama, 2025-05-12:
**Summary**
The generated ELF file cannot be correctly parsed and recognized by segger [ozone](https://www.segger.com/products/development-tools/ozone-j-link-debugger/)

**To Reproduce**
1. build a project
2. load the ELF file with ozone

**Expected behavior**
If the ELF is correctly parsed and recognized by ozone, there should be an `inline asm view` when click the `+` button

[image]

**Actual behavior**
 Unable to use the `inline asm view` function
[image]

**Environment**
LLVM-ET-Arm-19.1.5-Windows-x86_64

mikisama, 2025-05-12:
I found that this is caused by generating the dwarf version of ELF, add the `-gdwarf-4` flags can fix this

- ELF generated by LLVM-ET-Arm-19.1.5-Windows-x86_64 **without**  `-gdwarf-4` flags
[image]

-  Other mainstream compilers (such as armclang) now default to generating dwarf version 4

Is it too aggressive to generate dwarf-5 by default? The current tools seem to have poor compatibility

smithp35, 2025-05-12:
Thanks for asking the question on Dwarf standard defaults.

Arm Toolchain is tracking upstream llvm so unless we make a downstream change we'll inherit the upstream default. LLVM switched in 2022 to match GCC11 [1]. There was a discussion in the community about that in [2]. As we expect the majority of people using open-source tools to be using open-source debuggers (LLDB and GDB) we don't think it is worth changing the default.

Armclang defaulted to DWARF 4 largely due to the debugger in Arm DS not having good support (https://developer.arm.com/documentation/ka004927/latest/) although I think it did improve in the 2023 version. As a component of Arm DS it made sense for armclang to synchronize the DWARF support.

The Arm DS debugger has a specific error message:
```
Section .debug_info offset 0x10dd00: DWARF version 5 is not supported, try recompiling with DWARF4
```
Maybe worth asking Segger to put something like that in to give the user a hint that they need to recopmile using DWARF 4?

[1] https://github.com/llvm/llvm-project/commit/d3b26dea16108c427b19b5480c9edc76edf8f5b4
[2] https://discourse.llvm.org/t/dwarfv5-by-default-in-clang/59214

closed 2025-05-13 by mikisama