arm/arm-toolchain #337: [ATfE] toolchain build fails on runtimes on macOS

djazo, 2025-05-19:
**Summary**
Runtimes build fails when load LLVMExports.cmake

**To Reproduce**

cmake -G Ninja ..
ninja llvm-toolchain

**Expected behavior**
Build should complete.

**Actual behavior**
[14/1028] Performing compiler_rt-configure step for 'runtimes-aarch64a_be_soft_nofp_exn_rtti'
FAILED: /Users/azo/Code/arm-toolchain/arm-software/embedded/build/multilib-builds/runtimes/aarch64a_be_soft_nofp_exn_rtti/stamp/runtimes-aarch64a_be_soft_nofp_exn_rtti-compile
r_rt-configure
cd /Users/azo/Code/arm-toolchain/arm-software/embedded/build/multilib-builds/multilib/picolibc-build && /opt/homebrew/bin/cmake --build /Users/azo/Code/arm-toolchain/arm-softw
are/embedded/build/multilib-builds/runtimes/aarch64a_be_soft_nofp_exn_rtti/build --target compiler_rt-configure && /opt/homebrew/bin/cmake -E touch /Users/azo/Code/arm-toolcha
in/arm-software/embedded/build/multilib-builds/runtimes/aarch64a_be_soft_nofp_exn_rtti/stamp/runtimes-aarch64a_be_soft_nofp_exn_rtti-compiler_rt-configure
[0/1] Performing configure step for 'compiler_rt'
-- Could NOT find LibEdit (missing: LibEdit_INCLUDE_DIRS LibEdit_LIBRARIES)
CMake Error at /Users/azo/Code/arm-toolchain/arm-software/embedded/build/llvm/lib/cmake/llvm/LLVMExports.cmake:772 (add_library):
  ADD_LIBRARY called with SHARED option but the target platform does not
  support dynamic linking.
Call Stack (most recent call first):
  /Users/azo/Code/arm-toolchain/arm-software/embedded/build/llvm/lib/cmake/llvm/LLVMConfig.cmake:262 (include)
  cmake/Modules/CompilerRTUtils.cmake:312 (find_package)
  CMakeLists.txt:110 (load_llvm_config)
-- Configuring incomplete, errors occurred!

**Environment**
macOS Sequoia, homebrewed cmake, meson, ninja, python, qemu
both with AppleClang 17.0.0 and homebrewed clang 20.1.5

djazo, 2025-05-20:
Tested building on linux x86_64 and macOS arm64, fail comes from CMake 4.0.2.

voltur01, 2025-05-29:
Actually, the docs change said:

> CMake versions _older_ than 3.28 are not supported

So these are unrelated, there is a change in CMake logic in version 4 that breaks some of the complicated internal LLVM dependencies - we are actively looking into fixing this now.

PR #396 (2025-06-23): [ATfE] Use the runtimes project to build compiler-rt
When building the compiler-rt project using its standalone mode, its
CMake files will load the data from the pre-existing LLVM build via
`find_package(LLVM)` to collect information about the build
configuration. Doing this has the side effect of also including all the
libraries used by the pre-existing LLVM build as CMake targets of the
compiler-rt sub-project.

As the compiler-rt variants are being cross-compiled as part of ATfE to
target embedded environments, loading the library targets pre-compiled
to target the host environment during t [... cut]