arm/arm-toolchain #398: [ATFL] minimum RH/Rcoky version?

tonycurtis, 2025-06-23:
Wanted to test new compilers on a64fx cluster but we have Rocky 8.4.  GLIBC symbol errors.

Which version is required for atfl ?

pawosm-arm, 2025-06-24:
NB, Rocky Linux 8.4 was released on June 21, 2021. We don't have such old systems here anymore. RHEL8.10 is more than a year old now and so is the Rocky Linux 8.10. If possible, would you consider upgrading it?

pawosm-arm, 2025-06-24:
Ah no, this is experimental version of ATfL, prepared before this repository was ever created. It has been done last year on Ubuntu. The proper release is not distributed that way. See the official page: https://developer.arm.com/Tools%20and%20Software/Arm%20Toolchain%20for%20Linux there is deliberately no Download section in there. Instead, it leads you to this two pages:

https://developer.arm.com/documentation/110477/201/Welcome-to-the-Arm-Toolchain-For-Linux/Installation/Pre-installation-step - a set of instructions how to register our package repository in your system

https://developer.arm.com/documentation/110477/201/Welcome-to-the-Arm-Toolchain-For-Linux/Installation/Installation-step - how to install the toolchain from the registered repository

> (Installing as ordinary user)

Do you have `sudo` on this system? The instructions referred above need to be run as root.

tonycurtis, 2025-06-24:
Thanks for all the answers and pointers.  This isn't critical, just something I wanted to track, so it's not a big deal.  Will close.

tonycurtis, 2025-06-24:
I was going off of this post: [https://community.arm.com/arm-community-blogs/b/servers-and-cloud-computing-blog/posts/arm-performance-libraries-25-04-and-arm-toolchain-for-linux-20-release](https://community.arm.com/arm-community-blogs/b/servers-and-cloud-computing-blog/posts/arm-performance-libraries-25-04-and-arm-toolchain-for-linux-20-release).

(I might go and build it all)