arm/arm-toolchain #405: [Downstream Change]

nasherm, 2025-07-01:
**What is the change**
Patch to disable memcpy LDM/STM inlining for Cortex-V7M

**Why this change cannot be done upstream**
An attempt was made to upstream this patch. Upstream reviewers wanted a better solution that could be more fine-grained for the issue at hand.

pawosm-arm, 2025-07-01:
What is the link to the upstream PR this issue mentions?

dcandler, 2025-07-01:
> Upstream would prefer a different solution
And why can't we implement their preferred solution? We should be avoiding downstream changes if at all possible, and if it's not possible we should record the rationale.

Also, please don't close the issue, as it needs to track the change while it exists.