arm/arm-toolchain #446: [Downstream Change] Extend the Multilib system to support an IncludeDirs field via downstream patching.

simpal01, 2025-07-29:
**What is the change**
This patch extends the Multilib Yaml format to allow specifying an IncludeDirs for the header path/s per multilib variant. The goal is to enable fine-grained control over header search paths for each multilib variant. This feature is especially useful in setups where header paths deviate from the default bare-metal assumptions. For example, when headers are shared across target triples, it becomes necessary to organise them under target-triple-specific directories to ensure correct resolution.

**Why this change cannot be done upstream**
This is already in upstream review https://github.com/llvm/llvm-project/pull/146651. Since the review is pending, a downstream patch is made to enable IncludeDirs support in multilib.yaml

christophruethingbmw, 2026-10-20:
Hi @simpal01 do we have a plan on how to proceed here? The last comment in the upstream PR was

> Closing this PR since we decided to go with the hardcoded logic at the time, as outlined in the RFC.

So do we want to continue maintaining a deviation with the IncludeDirs or can we find a solution based on the upstream discussion?

In case you wonder why I stumbled across this: When using the upstream clangd with a generated compile commands file, I get a lot of errors regarding the not supported IncludeDirs resulting in standard include resolution does not properly work.

voltur01, 2026-10-20:
Hi @christophruethingbmw, thank you for reaching out! Let me comment while Simi is on holidays.

We did try to remove the downstream patch, but it turned out that the toolchain has places where it uses different triple names, thus broke our tests, see https://github.com/arm/arm-toolchain/pull/968 for the revert and details.

The strategy remains to remove this downstream change, however since it needs a bit more of upstream work we will need to get back to this.

BTW, as of ATfE 23 clangd should be part of the ATfE package, see https://github.com/arm/arm-toolchain/pull/844

christophruethingbmw, 2026-10-20:
@voltur01 , thanks you for the quick response, I checked #968 briefly but didn't dig into too much detail ;) Thank you for following up on this.